mcbittech/souliss-smart-thermostat-WiFi #3: Link non valido allo schema

dariocdj, 2015-12-27:
Ciao, 
https://github.com/mcbittech/souliss-smart-thermostat-WiFi/blob/master/SST_Connection.png?raw=true
non si vede più....
Mi sta arrivando il materiale....
grazie

Dario

mcbittech, 2015-12-27:
Closed #3 with commit 544596e

closed 2015-12-27 by mcbittech

mcbittech, 2015-12-27:
Closed #3
<https://github.com/mcbittech/souliss-smart-thermostat-WiFi/issues/3> with
commit 544596e
<https://github.com/mcbittech/souliss-smart-thermostat-WiFi/commit/544596eebb0ce489a9f942bdad8f74ab6ae5b138>

dariocdj, 2015-12-27:
> Nessun virus nel messaggio.
> Controllato da AVG - www.avg.com <http://www.avg.com>
> Versione: 2016.0.7294 / Database dei virus: 4489/11265 - Data di
> rilascio: 27/12/2015

grazie

-- 

-------------------------------------------------------------------
                    Dario ??Cdj??

      On Email: cdj@deskservice.com
      On Web  : http://www.cdj.deskservice.com
      On Irc  : Undernet.org #altraitalia
      On ICQ  : Uin # 3986985
      On MSN  : dariocdj@hotmail.it
      On Skype: dariocdj	

                Send me a fast reply !!!
------------------------------------------------------------------