mcbittech/souliss-smart-thermostat-WiFi #12: OH continuous refresh on "off" state

dariocdj, 2016-03-25:
Today i 've found my sst in "off" position correctly but in oh logs:

14:35:35.935 [DEBUG] [.b.s.i.n.udp.UDPSoulissDecoder:80   ] - updateTIMESTAMP:  TIMESTAMP_Nodo12 ( 153) = 2016-03-21T14:35:35.935CET
14:35:35.938 [DEBUG] [.s.i.n.t.SoulissGenericTypical:100  ] - Update State. Name: power, Typ: 0x11, Node: 12, Slot: 0. New State: 0.0
14:35:35.951 [DEBUG] [.s.i.n.t.SoulissGenericTypical:100  ] - Update State. Name: heating, Typ: 0x13, Node: 12, Slot: 0. New State: 0.0
14:35:35.955 [DEBUG] [.s.i.n.t.SoulissGenericTypical:100  ] - Update State. Name: cooling, Typ: 0x13, Node: 12, Slot: 0. New State: 0.0
14:35:35.957 [DEBUG] [.s.i.n.t.SoulissGenericTypical:100  ] - Update State. Name: fan_low, Typ: 0x11, Node: 12, Slot: 0. New State: 0.0
14:35:35.961 [DEBUG] [.s.i.n.t.SoulissGenericTypical:100  ] - Update State. Name: fan_med, Typ: 0x11, Node: 12, Slot: 0. New State: 0.0
14:35:35.970 [DEBUG] [.s.i.n.t.SoulissGenericTypical:100  ] - Update State. Name: null, Typ: 0x11, Node: 12, Slot: 0. New State: 0.0
14:35:35.974 [DEBUG] [.s.i.n.t.SoulissGenericTypical:100  ] - Update State. Name: fan_auto_mode, Typ: 0x11, Node: 12, Slot: 0. New State: 0.0
14:35:35.977 [DEBUG] [.s.i.n.t.SoulissGenericTypical:100  ] - Update State. Name: heating_cooling, Typ: 0x11, Node: 12, Slot: 0. New State: 1.0
14:35:35.980 [DEBUG] [.b.s.i.n.udp.UDPSoulissDecoder:300  ] - decodeStateRequest: power (49). Thermostat= 128, Temp.Measured= 20.796875, Temp.SetPoint= 20.796875
14:35:35.991 [DEBUG] [.o.b.s.i.n.udp.UDPServerThread:61   ] - Packet received

This happen continuously...

fazioa, 2016-06-02:
Dario, the issue still exists?

dariocdj, 2016-06-02:
No close it, tnx

closed 2016-06-02 by dariocdj