mcbittech/souliss-smart-thermostat-WiFi #13: Modbus support

marksev1, 2016-03-30:
Can you please add modbus support into this project?

mcbittech, 2016-04-01:
Souliss support modbus, but only in AVR mcu.
There isn't modbus support on ESP now.
If you need modbus, i can suggest you to use a gateway Avr based and retrieve data from your SST, in order to have modbus support.

Flavio.

marksev1, 2016-04-01:
Yeah thats not a good solution, to have redundant hardware just for that sole purpose. Hopefully you guys will provide also some other way to interact with esp outside of Souliss's internal protocols. Something widespread and easy to interact.

closed 2016-04-01 by marksev1

mcbittech, 2016-04-01:
Why you need modbus?

Flavio.

marksev1, 2016-04-02:
I don't neccessary need modbus, I just need some protocol other than souliss's, some protocol which is  easier to interact with and which is more widespread. So I can build my custom dashboard and software around it.

mcbittech, 2016-04-02:
SST is Souliss based :)
Se doesent plain external protocol.
You cannot build your dasboard around souliss?

Flavio.