mcbittech/souliss-smart-thermostat-WiFi #34: Files compiled with errors

igbor, 2016-11-01:
Files compiled with errors. Established Arduino AVR v1.6.11 ESP 8266 v2.3.0 Souliss v.7.2.1

C:\Users\igbor\Documents\Arduino\e05_ESP8266_SST_webconf\e05_ESP8266_SST_webconf.ino: In function 'void subscribeTopics()':

e05_ESP8266_SST_webconf:114: error: 'subscribedata' was not declared in this scope
   if (subscribedata(TOPIC1, mypayload, &mypayload_len)) {
                                                                                          ^
C:\Users\igbor\Documents\Arduino\e05_ESP8266_SST_webconf\e05_ESP8266_SST_webconf.ino: In function 'void publishHeating_ON_OFF()':

e05_ESP8266_SST_webconf:248: error: 'publishdata' was not declared in this scope
     publishdata(SST_HEAT_ONOFF, &HEAT_ON, 1);
                                                                            ^
e05_ESP8266_SST_webconf:250: error: 'publishdata' was not declared in this scope
     publishdata(SST_HEAT_ONOFF, &HEAT_OFF, 1);
                                                                              ^
exit status 1
'subscribedata' was not declared in this scope

fazioa, 2016-11-01:
In last Souliss release subscribedata as changed in sbscrb and publish
in pblsh

Look at .\souliss\base\PublishSubscribe.h

Tonino Fazio

mcbittech, 2016-11-08:
@igbor have you solved?

igbor, 2016-11-09:
Thank you for all compiles fine, the author has made changes. Upgrade to github

closed 2016-11-09 by mcbittech