mcbittech/souliss-smart-thermostat-WiFi #38: ESP-12E Random Soft WDT Resets

brunodymios, 2017-01-04:
Hi there,
I'm really interested in this project and I'm testing it on a NodeMCU ESP-12E development board.
I randomly receive software WDT resets. It can happen right after 10 minutes or a couple of hours, it is not sistematic. It happens both when powered with USB and with an external power supply (7805 connected to the Vin pin of the board).

Here you are the stacktrace I'm receiveing in the serial monitor window:

```
Refresh Circle White 
Refresh Circle White OK 
Refresh Circle Black 
Refresh Circle Black OK 
Refresh HOME Screen 

Soft WDT reset

ctx: cont 
sp: 3fff1640 end: 3fff1bd0 offset: 01b0

>>>stack>>>
3fff17f0:  3fff0a40 3fff1810 00000029 40215609  
3fff1800:  00000000 3fff1ad8 3fff1ab8 4021a54b  
3fff1810:  00000000 0000000e 00000005 00000030  
3fff1820:  00000001 3fff1ad8 3fff1ab8 4021a864  
3fff1830:  40215482 63045964 00000020 401076bc  
3fff1840:  3fffc278 3fff1ad8 3fff1ab8 4021a8f3  
3fff1850:  402154b1 00000030 3fff1ab8 4021ba14  
3fff1860:  4021549b 3fff1ad8 3fff1ab8 4021bced  
3fff1870:  00000025 0000005a 0000fff8 fffffff8  
3fff1880:  000000b4 000000b4 0000004a 00000015  
3fff1890:  4021549b 00000030 00000008 0000003b  
3fff18a0:  00000025 00000039 3fff1ab8 40247f5d  
3fff18b0:  fc0000ff 402154e0 402154e0 4021b27b  
3fff18c0:  000000a4 00000055 27f40000 4021ad68  
3fff18d0:  00000000 0000003b 00000000 fffffffc  
3fff18e0:  0000003b 00000000 00000025 401076bc  
3fff18f0:  4021a8a4 0000000f 3fffc230 0000003f  
3fff1900:  00000000 3fff1ab8 3fff1b34 4021ae4d  
3fff1910:  00000025 0000003e ffffffcf 00000000  
3fff1920:  4021b97c 00000000 000000ff 3fff1ab8  
3fff1930:  fffffffc 00000000 3fff1ab8 4021af03  
3fff1940:  3fff1ab8 00000050 00000000 4021b000  
3fff1950:  00000031 00000001 3fff1ab8 40219d28  
3fff1960:  00009b98 00000000 3fff1ab8 3fff19f0  
3fff1970:  3fff1ab0 00000000 3fff1ab0 40215685  
3fff1980:  000000c5 00000002 3fff19ee 4021892c  
3fff1990:  4000df64 00000030 00000011 3fff1ab0  
3fff19a0:  00000031 0000000a 3fff1ab0 402160f1  
3fff19b0:  3fff19ee 3fff1a30 3fff1ab8 4021a8f3  
3fff19c0:  000000a3 3fff1a20 3fff19ed 402162ad  
3fff19d0:  3fff00f0 3fff1a30 3fff1ab8 4021bd51  
3fff19e0:  00000000 3fff1a20 3fff1ab8 3831b9b1  
3fff19f0:  3fff1c00 3fff1aba 3fff1ab8 00000009  
3fff1a00:  00000009 3fff1ab8 3fff1ab8 00000009  
3fff1a10:  3fff1c18 3fff1ab8 3fff1ab8 00000009  
3fff1a20:  3fff1c18 00000012 3fff1ab0 4021630b  
3fff1a30:  00000000 014000f0 3fff1ab8 40247f29  
3fff1a40:  3fff1c18 3fff1ab8 3fff1ab0 40216324  
3fff1a50:  41973333 41973333 3fff1ab0 4020a751  
3fff1a60:  00000012 41a00000 4288cccd 402161ec  
3fff1a70:  00000000 3fff1c18 3fff1c18 40216238  
3fff1a80:  3fff0226 3fff1c18 3fff1ab8 3fff0ba4  
3fff1a90:  00000000 3fff0224 3fff00f0 4020ebc3  
3fff1aa0:  40000f83 00000030 00000020 401076bc  
3fff1ab0:  3ffe89b0 00000000 00a00001 00000078  
3fff1ac0:  4021bba8 4021a880 4021a8a4 014000f0  
3fff1ad0:  4021b97c 402154e0 0000003b 00000000  
3fff1ae0:  00010042 00010008 00000000 ff660000  
3fff1af0:  00000000 00000000 00000000 00000000  
3fff1b00:  00000000 00000000 00000000 00000000  
3fff1b10:  00000000 00000000 00000000 00000000  
3fff1b20:  00000000 00000000 014000f0 4024e9e4  
3fff1b30:  4021a9dc 4024ef11 001ffffc 3e253a1f  
3fff1b40:  00000001 0506025f 05010606 003e2507  
3fff1b50:  30f42ef3 098400f3 000014de 00000000  
3fff1b60:  02fff42e ffffff0f ffffffff 07ffffff  
3fff1b70:  0000010d 4021a8a4 4021a880 0050fffc  
3fff1b80:  feefef00 3fff2b34 3fff267c 00000000  
3fff1b90:  00000000 00000000 feefeffe feefeffe  
3fff1ba0:  00000000 00000000 00000001 3fff0ba4  
3fff1bb0:  3fffdad0 00000000 3fff0b9c 4021588c  
3fff1bc0:  feefeffe feefeffe 3fff0bb0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
ÿ

```
Addresses shown before the stack trace are not fixed, they change between subsequent crashes. Here's another crash log, right after the first one:

```
Refresh Circle Green 
Refresh Circle Green OK 

Soft WDT reset

ctx: cont 
sp: 3fff16c0 end: 3fff1bd0 offset: 01b0

>>>stack>>>
```

At the beginning of the code, in the setup() I've added some additional debug info:
```
Heap: 37424
Boot Vers: 31
CPU: 80
SDK: 1.5.3(aec24ac9)
Chip ID: 422598
Flash ID: 1458400
Flash Size: 4194304
Vcc: 65535
```

Boot mode (3,6) seems to be correct: booting from SPI flash memory, while the reset cause (2) means that there is something with the RST pin. I checked the connections and they are OK and I added a 10k pull-up resistor on the GPIO 0 pin, with no luck.

Anyone facing this problem? Any hint to solve it?

Thank You

mcbittech, 2017-01-04:
Hello @brunodymios ,
what kind of release are you working on? your general hardware and schematics?
Are you uploading your code following wiki?
https://github.com/mcbittech/souliss-smart-thermostat-WiFi/wiki/SST---WiFi-Souliss-Smart-Thermostat-Wiki

Flavio

brunodymios, 2017-01-04:
Hi @mcbittech, thanks for your answer.
Yes, I followed the wiki you linked me to setup the firmware (branch "friariello" for souliss and this repo for the thermostat-specific code). I'm using Arduino IDE **1.6.5** with ESP8266 board version **2.3.0** installed. 

I'm using a **NodeMCU** board found on banggood. [Here's the link](http://www.banggood.com/it/NodeMcu-Lua-WIFI-Internet-Things-Development-Board-Based-ESP8266-CP2102-Wireless-Module-p-1097112.html?rmmds=search). 
The LCD is similar to your one [Link](http://www.banggood.com/it/2_2-Inch-Serial-TFT-SPI-LCD-Screen-Module-HD-240-x-320-5110-Compatible-p-912854.html?rmmds=search).
DHT22 is used for Temp/Humidity sensor.

I didn't connect any relay for the moment in order to minimize the complexity of the schematic for testing. I'm using a simple resistor + led to simulate it. All the schematic is built on a breadboard to prototype.

The device is generally working well: temperature and humidity correctly displayed, user menu, set point, etc. Reboots seems to be really random, always with same reason (2 - rst pin and boot mode 3,6) but stack traces point to different memory locations.

Bruno

mcbittech, 2017-01-04:
@brunodymios 
Can you try with last IDE supported 1.6.12 please, with [this ](https://github.com/mcbittech/souliss-smart-thermostat-WiFi/releases/tag/SST_v1.2_prewbs_STABLE)version of SST?
Your rst pin are floating or connected?
We haven't tried with this ESP board, if this tests doesen't work, we must go deeper inside hardware/pinout of your board. In this case please attach a schematics of your board.

Flavio

brunodymios, 2017-01-04:
@mcbittech
I've tried right now. I updated the IDE to the 1.6.12 version and I used the version of SST you indicated me (including libraries). I added a 10k pull-up resistor to the RST and GPIO0 pins.

I received a reboot after a couple of minutes the following stack trace:
```
>>>stack>>>
3fff18c0:  00000000 00000000 3fff1b98 40219b64  
3fff18d0:  3fff0a30 3fff18f0 00000008 402145f3  
3fff18e0:  00000000 3fff1bb8 3fff1b98 4021a387  
3fff18f0:  3f00ff66 40101933 04000000 00000001  
3fff1900:  00000001 3fff1bb8 3fff1b98 4021a6a0  
3fff1910:  00000000 00000000 3fffc230 4000050c  
3fff1920:  4021ace4 3fff1bb8 3fff1b98 4021a72f  
3fff1930:  4021ad35 4024e9fc 3fff1b98 4021b850  
3fff1940:  00000000 3fff1bb8 3fff1b98 4021bb29  
3fff1950:  00000007 00000034 0000002a 0000002a  
3fff1960:  00000068 00000068 0000000e 00000015  
3fff1970:  00000000 3fff1b98 00000020 00000015  
3fff1980:  0000000c 00000007 3fff1b98 40247d45  
3fff1990:  4021a665 00000030 3fffc230 4021b0b7  
3fff19a0:  40219ed8 00000030 0000001d 4021aba4  
3fff19b0:  00000019 00000015 00000001 00000015  
3fff19c0:  00000015 00000000 00000007 00000066  
3fff19d0:  00000000 00000085 00000000 0000001e  
3fff19e0:  00000007 3fff1b98 3fff1c14 4021ac92  
3fff19f0:  00000025 0000003e ffffffcf 00000000  
3fff1a00:  00000000 00000007 00000055 3fff1b98  
3fff1a10:  fffffffc 00000000 3fff1b98 4021ad3f  
3fff1a20:  0000006e 00000050 00000000 4021ae3c  
3fff1a30:  00000032 00000001 3fff1b98 40219b64  
3fff1a40:  00009b10 00000000 3fff1b98 3fff1ad0  
3fff1a50:  3fff1b90 00000000 3fff1b90 40214679  
3fff1a60:  000000c5 00000002 3fff1ace 402187a4  
3fff1a70:  3fff0a30 00000001 3fff1b98 3fff1b90  
3fff1a80:  00000032 0000000a 3fff1b90 40215d1d  
3fff1a90:  3fff1ace 3fff1b10 3fff1b98 4021a72f  
3fff1aa0:  00000000 3fff1b00 3fff1acd 40215ed9  
3fff1ab0:  00000009 3fff1b10 3fff1b98 4021bb8d  
3fff1ac0:  00000000 3fff1b00 3fff1b98 3132b7ed  
3fff1ad0:  3fff0b00 3fff1b9a 3fff1b98 00000004  
3fff1ae0:  00000009 3fff1b98 3fff1b98 00000004  
3fff1af0:  3fff0bc0 3fff1b98 3fff1b98 00000004  
3fff1b00:  3fff0bc0 00000015 3fff1b90 40215f37  
3fff1b10:  00000000 014000f0 3fff1b98 40247d11  
3fff1b20:  3fff0bc0 3fff1b98 3fff1b90 40215f50  
3fff1b30:  41ab3333 41ab3333 3fff1b90 4020a751  
3fff1b40:  00000015 41a00000 4271999a 40215e18  
3fff1b50:  00000000 3fff0bc0 3fff0bc0 40215e64  
3fff1b60:  40217cd8 3fff0bc0 3fff1b98 3fff0c7c  
3fff1b70:  00000000 3fff0224 3fff00f0 4020eae7  
3fff1b80:  40000f83 00000030 0000000e ffffffff  
3fff1b90:  3ffe89b0 00000000 00a00001 00000078  
3fff1ba0:  4021b9e4 4021a6bc 4021a6e0 014000f0  
3fff1bb0:  4021b7b8 402144d4 002a0087 0000ff66  
3fff1bc0:  0001000e 00010000 00000000 ff660000  
3fff1bd0:  00000000 00000000 00000000 00000000  
3fff1be0:  00000000 00000000 00000000 00000000  
3fff1bf0:  00000000 00000000 00000000 00000000  
3fff1c00:  00000000 00000000 014000f0 4024e7cc  
3fff1c10:  4021a818 4024ed21 001ffffc 3e251519  
3fff1c20:  00000004 0506025f 05010606 003e2507  
3fff1c30:  30f42ef3 098400f3 000014de 00000000  
3fff1c40:  02fff42e ffffff0f ffffffff 07ffffff  
3fff1c50:  0000010d 4021a6e0 4021a6bc 0050fffc  
3fff1c60:  feefef00 3fff2b3c 3fff2684 00000000  
3fff1c70:  00000000 00000000 feefeffe feefeffe  
3fff1c80:  00000000 00000000 00000001 3fff0c7c  
3fff1c90:  3fffdad0 00000000 3fff0c74 40216e30  
3fff1ca0:  feefeffe feefeffe 3fff0c90 40100718  
<<<stack<<<
[... 32 lines cut ...]
0x40215f37: Print::print(long, int) at C:\Users\bambr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 76
0x40247d11: ucg_GetDimension at C:\Program Files (x86)\Arduino\libraries\Ucglib\utility/ucg_dev_msg_api.c line 94
0x40215f50: Print::print(int, int) at C:\Users\bambr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 76
0x4020a751: Ucglib::undoScale() at C:\Program Files (x86)\Arduino\libraries\Ucglib/Ucglib.h line 124 (discriminator 1)
:  (inlined by) display_layout2_HomeScreen(Ucglib_ILI9341_18x240x320_HWSPI, float, float, float) at C:\Users\bambr\AppData\Local\Temp\arduino_build_772061\sketch/display2.cpp line 205 (discriminator 1)
0x40215e18: Print::println() at C:\Users\bambr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 76
0x40215e64: Print::println(char const*) at C:\Users\bambr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 76
0x40217cd8: run_scheduled_functions() at C:\Users\bambr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Schedule.cpp line 87
0x4020eae7: loop at C:\Program Files (x86)\Arduino\libraries\souliss-friariello/base/T2n.cpp line 337
0x4021b9e4: ucg_dev_scale2x2 at C:\Program Files (x86)\Arduino\libraries\Ucglib\utility/ucg_scale.c line 97
0x4021a6bc: ucg_ext_ili9341_18 at C:\Program Files (x86)\Arduino\libraries\Ucglib\utility/ucg_dev_ic_ili9341.c line 373
0x4021a6e0: ucg_dev_ili9341_18x240x320 at C:\Program Files (x86)\Arduino\libraries\Ucglib\utility/ucg_dev_tft_240x320_ili9341.c line 107
0x4021b7b8: ucg_dev_rotate90 at C:\Program Files (x86)\Arduino\libraries\Ucglib\utility/ucg_rotate.c line 69
0x402144d4: ucg_com_arduino_4wire_HW_SPI at C:\Program Files (x86)\Arduino\libraries\Ucglib/Ucglib.cpp line 1038
0x4024e7cc: ucg_DrawL90SEWithArg at ?? line ?
0x4021a818: ucg_font_calc_vref_font at C:\Program Files (x86)\Arduino\libraries\Ucglib\utility/ucg_font.c line 918
0x4024ed21: ucg_DrawL90SEWithArg at ?? line ?
[... cut ...]
0x40216e30: loop_wrapper at C:\Users\bambr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\bambr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109
```
It seems to be something related to the display. I think my LCD is using a **ILI9340C** controller instead of the **ILI9341**. Can this be an issue in your opinion? I did not see any additional issue with GUI or display except those random reboots.

Thank You

mcbittech, 2017-01-04:
@brunodymios 
try to add some yield(); from [line](https://github.com/mcbittech/souliss-smart-thermostat-WiFi/blob/master/e05_ESP8266_SST_webconf/e05_ESP8266_SST_webconf.ino#L770)

something like this:
**yield();**
            display_layout2_print_circle_white(ucg);
**yield();**
            display_layout2_print_circle_black(ucg);
**yield();**
            display_layout2_HomeScreen(ucg, temperature, humidity, setpoint);
**yield();**
            display_layout2_print_datetime(ucg);
**yield();**

Then try to move your display pin to other free pins, pin like 2 and 15 to other position.

Let me now

Flavio

brunodymios, 2017-01-05:
Hi @mcbittech
following your indication I added the yield instructions and the device was up and running from 4pm yesterday to 11am this morning without reboots.

As a final test, I flashed again the version without the yield instructions and I saw again a reboot in about 20 minutes.

What do you think? Can we consider this issue closed?

Thank You
Bruno

mcbittech, 2017-01-05:
@brunodymios i don't think this is an issue related to SST code, but i think this is an ESP core issue.
for the moment i commit new code changes on SST side.
Keep your SST on test and let me now please 👍 
For the moment we keep open this issue for some days.

Flavio

brunodymios, 2017-01-06:
Completely agree with you, this is not an issue of the SST but rather an improvement of it to take into account this issue in the ESP core.
I'll leave the SST on for testing and I'll let you know.

Thanks.
Bruno

mcbittech, 2017-02-04:
@brunodymios have any news?

Thanks
Flavio

brunodymios, 2017-02-25:
Did not face the issue again 👍

mcbittech, 2017-11-22:
solved

closed 2017-11-22 by mcbittech