mcbittech/souliss-smart-thermostat-WiFi #42: Sospensione crono

fazioa, 2017-02-03:
Sospendere il crono per 30 minuti quando si imposta manualmente il setpoint

mcbittech, 2017-02-04:
Non sono del tutto d'accordo.
In ogni caso se vuoi testarlo è molto semplice, basta triggerare la chiamata del crono ogni tempo che vuoi.
Se non ricordo male basta portare: 
      if (getCrono()) {
        Serial.println("CRONO: aggiornamento");
        setSetpoint(checkNTPcrono(ucg));
        setEncoderValue(checkNTPcrono(ucg));
        Serial.print("CRONO: setpoint: "); Serial.println(setpoint);
      }
che adesso è chiamata in una slow_50s, in una slow_xx che puoi configurare a piacimento.

closed 2017-02-04 by mcbittech

dariocdj, 2017-12-11:
Io lo farei per 2/3 ore... Mezz'ora manco te ne accorgi... Pop-up di conferma necessario