mcbittech/souliss-smart-thermostat-WiFi #48: wbserver_beta non compila con DYNAMIC = 1

dpeddi, 2017-03-01:
Sembra non sia possibile a compilare wbserver_beta con DYNAMIC =1

questo perché si formano dei conflitti con ESPAsynWebServer e 
#include "conf/RuntimeGateway.h"            // This node is a Peer and can became a Gateway at runtime
#include "conf/DynamicAddressing.h"         // Use dynamically assigned addresses
#include "conf/WEBCONFinterface.h"          // Enable the WebConfig interface

occorre quindi capire se souliss passerà a ESPAsyncWebServer oppure se sarà necessario incorporare in SST una webconf  basata su ESPAsynWebServer, più o meno come ha fatto gmag11(https://github.com/gmag11/FSBrowserNG) e perché no salvare su spiffs anche la configurazione del wifi.

Trà l'altro (se ci si stà nella flash) potrebbe essere interessante includere l'editor ACE (potrebbe tornar comodo per modificare alcune configurazioni... avrei in mente di sfruttarlo per modificare l'abbinamento topic/espid senza doverlo cablare su preferences).. ma questo è un altro film...

Segue l'errore:
---
Compiling .pioenvs\e05_ESP8266_SST_webconf\lib\ESP8266WebServer\ESP8266WebServer.o
In file included from D:/Apps/Atom_x64/projects/souliss-smart-thermostat-WiFi_wbserver_beta/e05_ESP8266_SST_webconf/e05_ESP8266_SST_webconf.ino:65:0:
D:\Apps\Atom_x64\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer.h:30:19: error: redeclaration of 'HTTP_ANY'
enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

(souliss\interfaces\mcu_esp8266\webconfig\webconfig.cpp ESP8266WebServer server(HTTP_PORT); )

e

In file included from .piolibdeps\souliss/Souliss.h:168:0,
from D:/Apps/Atom_x64/projects/souliss-smart-thermostat-WiFi_wbserver_beta/e05_ESP8266_SST_webconf/e05_ESP8266_SST_webconf.ino:89:
.piolibdeps\souliss/tools/store/store.cpp: In function 'void Store_Init()':
.piolibdeps\souliss/tools/store/store.cpp:37:3: error: 'EEPROM' was not declared in this scope

EEPROM.begin(STORE__SIZE);
^

per lo stesso motivo

dpeddi, 2017-04-22:
prerequisito: creata pull request https://github.com/souliss/souliss/pull/292

dariocdj, 2017-12-09:
credo che hai già mergiato la mia modifiche dove extern web server lo fa indipendentemente se dinamic=1 o no.....

mcbittech, 2017-12-11:
ok, chiudo perchè era vecchio, in attesa di trasferimento