mcbittech/souliss-smart-thermostat-WiFi #54: Stampa aggiornamento setpoint se sono nelle pagine di topic

pacioc193, 2017-12-07:
Se viene cambiato il setpoint mentre si è nel pannello di pagine topic viene stampato a schermo il setpoint.

pacioc193, 2017-12-07:
SHIFT_50ms(0) {
      //set point attuale
      setpoint = Souliss_SinglePrecisionFloating(memory_map + MaCaco_OUT_s + SLOT_THERMOSTAT + 3);
      //Stampa il setpoint solo se il valore dell'encoder è diverso da quello impostato nel T31
      switch (SSTPage.actualPage) {
        case PAGE_MENU :
          if (!getMenuEnabled()) {
            SSTPage.actualPage = PAGE_HOME;
            initScreen();
            setUIChanged();
          }
          break;
        case PAGE_CRONO :
          break;
        case PAGE_HOME:
          if (arrotonda(getEncoderValue()) != arrotonda(encoderValue_prec) && !bChildLock) {
            FADE = 1;
            //TICK TIMER
            timerDisplay_setpoint_Tick();
            //SETPOINT PAGE ////////////////////////////////////////////////////////////////

            if (getLayout1()) {
              SERIAL_OUT.println("display_setpointPage - layout 1");
              display_layout1_background(ucg, arrotonda(getEncoderValue()) - arrotonda(setpoint));

              display_layout1_setpointPage(ucg, getEncoderValue(), temperature, humidity, getSoulissSystemState());
            }
            else if (getLayout2()) {
              SERIAL_OUT.println("display_setpointPage - layout 2");
              display_layout2_Setpoint(ucg, getEncoderValue(), getSoulissSystemState(), bChildLock);
            }
          }
          encoderValue_prec = getEncoderValue();
          break;
        default:
          break;
      }
    }

pacioc193, 2017-12-07:
` SHIFT_50ms(0) {
      //set point attuale
      setpoint = Souliss_SinglePrecisionFloating(memory_map + MaCaco_OUT_s + SLOT_THERMOSTAT + 3);
      //Stampa il setpoint solo se il valore dell'encoder è diverso da quello impostato nel T31
      switch (SSTPage.actualPage) {
        case PAGE_MENU :
          if (!getMenuEnabled()) {
            SSTPage.actualPage = PAGE_HOME;
            initScreen();
            setUIChanged();
          }
          break;
        case PAGE_CRONO :
          break;
        case PAGE_HOME:
          if (arrotonda(getEncoderValue()) != arrotonda(encoderValue_prec) && !bChildLock) {
            FADE = 1;
            //TICK TIMER
            timerDisplay_setpoint_Tick();
            //SETPOINT PAGE ////////////////////////////////////////////////////////////////

            if (getLayout1()) {
              SERIAL_OUT.println("display_setpointPage - layout 1");
              display_layout1_background(ucg, arrotonda(getEncoderValue()) - arrotonda(setpoint));

              display_layout1_setpointPage(ucg, getEncoderValue(), temperature, humidity, getSoulissSystemState());
            }
            else if (getLayout2()) {
              SERIAL_OUT.println("display_setpointPage - layout 2");
              display_layout2_Setpoint(ucg, getEncoderValue(), getSoulissSystemState(), bChildLock);
            }
          }
          encoderValue_prec = getEncoderValue();
          break;
        default:
          break;
      }
    }`

pacioc193, 2017-12-07:
case PAGE_HOME:
          if (arrotonda(getEncoderValue()) != arrotonda(encoderValue_prec) && !bChildLock) {
            FADE = 1;
            //TICK TIMER
            timerDisplay_setpoint_Tick();
            //SETPOINT PAGE ////////////////////////////////////////////////////////////////

            if (getLayout1()) {
              SERIAL_OUT.println("display_setpointPage - layout 1");
              display_layout1_background(ucg, arrotonda(getEncoderValue()) - arrotonda(setpoint));

              display_layout1_setpointPage(ucg, getEncoderValue(), temperature, humidity, getSoulissSystemState());
            }
            else if (getLayout2()) {
              SERIAL_OUT.println("display_setpointPage - layout 2");
              display_layout2_Setpoint(ucg, getEncoderValue(), getSoulissSystemState(), bChildLock);
            }
          }
          encoderValue_prec = getEncoderValue();
          break;
        default:
          break;

pacioc193, 2017-12-10:
Questo è stato risolto quando hai mergiato la mia pull request

mcbittech, 2017-12-11:
Si, solo su branch webserver, lascio aperto fino al merge su master

mcbittech, 2017-12-11:
@dariocdj sarebbe apprezzato un test incrociato da parte tua, attendiamo esiti

mcbittech, 2017-12-11:
@pacioc193 era già aperto questo issue, vedi #16.
cerchiamo di fare attenzione e non fare duplicati se possibile.