mcbittech/souliss-smart-thermostat-WiFi #59: Calibration offset range

mcbittech, 2017-12-09:
Aumentare il range di calibrazione dell'offset della sonda di misura

pacioc193, 2017-12-12:
<html><!DOCTYPE html> 
<html lang="en"> 
<head> 
    
	<meta charset="utf-8">
  	<meta http-equiv="Content-Type" content="text/html; charset=utf-8" />
	<title>SST Setting</title>
	<link rel="stylesheet" href="css/jquery-ui.min.css">
	<script src="javascript/jquery.js"></script>
	<script src="javascript/jquery-ui.min.js"></script>
	<link href="https://fonts.googleapis.com/css?family=Comfortaa" rel="stylesheet"> 

	<meta id="viewport" name="viewport" content="initial-scale=1.0, minimum-scale=1.0, maximum-scale=1.0, user-scalable=no" />	
      <script>
      $(document).ready(function() {		
		$.ajaxSetup( { cache: false } );
		//importo file e divido i valori 
			$.getJSON('sst_settings.json', function(jd) {
				console.log (jd);
				if (jd.Orologio == 1) {
					$("#orologio").attr("checked", true);
					document.getElementById("orologio").checked="checked";
					}
					else{
					 $("#orologio").attr("checked", false);
					 }
				if (jd.CronoLearn == 1) {
					$("#cronolearn").attr("checked", true);
					document.getElementById("cronolearn").checked="checked";
					}
					else{
					 $("#cronolearn").attr("checked", false);
					 }
				if (jd.Crono == 1) {
					$("#crono").attr("checked", true);
					document.getElementById("crono").checked="checked";
					}
					else {
					$("#crono").attr("checked", false);
					}
				if (jd.Dispositivo == 1) {
					$("#dispositivo").attr("checked", true);
					document.getElementById("dispositivo").checked="checked";
					}
					else {
					$("#dispositivo").attr("checked", false);
					}
				if (jd.DayLightSavingTime == 1) {
					$("#orasolare").attr("checked", true);
					document.getElementById("orasolare").checked="checked";
					}
					else {
					$("#orasolare").attr('checked', false);
					}
				if (jd.Layout1 == 1){
					$("#layout1").attr("checked", true);
					document.getElementById("layout1").checked="checked";	
					$("#layout2").attr("checked", false);
					}
					
				if (jd.Layout2 == 1){
					$("#layout2").attr("checked", true);
					document.getElementById("layout2").checked="checked";
					$("#layout1").attr("checked", false);
					}
				document.getElementById ("temp").value = jd.Luminosita;
				document.getElementById ("offs").value = jd.OffsetDHT;		
				document.getElementById ("timezone").value=jd.Tzone;	
				$("#lumi").slider('value',jd.Luminosita);
				$("#offset").slider('value',jd.OffsetDHT);
[... 151 lines cut ...]
				<p id="lumi" style="width:200px; height:14px;font-family:Confortaa;"></p>
				<input type="text" id="temp" name="temp" readonly style="border:1; color:#ff9900; font-weight:bold; font-size:x-large; width: 2.5em; text-align:center" size="4">
			</fieldset>
		</tr>
		<tr>
			<fieldset>
			<legend align="center" style="font-family:Comfortaa;">Offset DHT</legend>
				<p id="offset" style="width:200px; height:14px;"></p>
				<input type="text" id="offs" name="offs" readonly style="border:1; color:#349900; font-family:Comfortaa; font-weight:bold; font-size:x-large; width:2.7em; text-align:center" size"5">
				
			</fieldset>
		</tr>
	</table>
      
      </div>
     
       <div align="center" style = " width:320px; background-position:center; margin:0 auto;" >
		<p></p>
		<input type="submit" value="Cancel" onclick="window.location = './index.htm';" /				>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;		
		<input type="submit" value="Salva" onclick="saveSchedule();" />
		<p></p>
       </div>
   </body>
	
</html>

pacioc193, 2017-12-12:
$("#offset").slider({
	  	orientation: "orizontal",
	  	range: "min",
	  	min: -10,
	  	max: 10,
	  	step: 0.1,
	  	value: document.getElementById ("offs").value,
	  	slide: function( event, ui ) {
			$( "#offs" ).val( ui.value);
		
			}
		});

Dentro il file setting.htm della spiffs ...